Ulm-IQO/qudi-legacy #537: Naming conflicts in interfaces - the case of get_constraints

alrik-durand, 2019-07-12:
## *What* is affected by this bug?
This is not a bug, this is a design problem. One modules can not implement two interfaces if they both have a get_constraints function.

## *When* does this occur?
When I tired using the TimeHarp 260 as both a slow and a fast counter.

## *Where* on the platform does it happen?
At the developping stage.

## Expected behavior (i.e. solution)
A module should be able to implement all the interfaces if he wants to, without having to use interfuses tricks.
I see two obvious possible solutions : 
- Renaming the conflicting function, for example get_constraints_SlowCounterInterface
- Adding a parameter to the get_constraints function that specifies the name of the interface.
Example
`
stuff = hardware.get_constraints('SlowCounterInterface')
`
Both these changes would require a lot of changes in critical parts of the Qudi project so it can not be donne slightly. That is why I would like you opinion before creating a PR.

Neverhorst, 2019-07-12:
This is a known issue.

The problem is right now being tackled in the following branch:
https://github.com/Ulm-IQO/qudi/tree/interface_method_decorator

I assume it will be ready to merge with master within the next two weeks.

alrik-durand, 2019-07-12:
Thank you for your answer, I had not seen this !

This looks very beautiful ! I will wait for it then.

closed 2019-07-12 by alrik-durand