Ulm-IQO/qudi-legacy #543: Installation from .yml throws error

TraceBivens, 2019-08-12:
When running `conda env create -f environment.yml ` conda throws ResolvePackageNotFound for the entire dependency list. Some digging showed that removing the build numbers from the .yml file significantly reduced the problem, but conda still can't find libgcc-ng 7.2.0 along with three other specified packages. I've tried to find the specified version with pip and had no luck. Advice?

drogenlied, 2019-08-13:
Yes, you could try to specify what exactly you are doing.

This starts with stating your operating system version, continues with the Anaconda version and install location and ends with the exact environment that you are trying to install.

Without that information, there is no need to even start guessing about what is going wrong.

It could also be that the conda package repository is broken again.

TraceBivens, 2019-08-13:
I was trying to get it running on a Mac at first on an apparently naive assumption that python stuff is typically cross-platform. After posting this I had the bright idea to try it on my Ubuntu machine and had no problems. Sorry about that 😅.

How exactly does one go about setting up the software to recognize hardware instruments and execute experiments? Qudi looks super cool but I haven’t been able to find anything in the documentation that talks about how to establish communication in the first place. Is that what the interpreter pane is for?

drogenlied, 2019-08-14:
Pure Python is cross-platform, all the binary parts from conda are typically not.

I know that people have run Qudi on OSX with handcrafted environments but nobody has bothered to send us their environment file or other configuration.

About using real hardware:

Basically, you have to write your own configuration file that corresponds to your setup.

Read the getting started page: https://ulm-iqo.github.io/qudi-generated-docs/html-docs/get-started.html

Read the documentation about the config file: 
https://ulm-iqo.github.io/qudi-generated-docs/html-docs/config-explanation.html

All the logic and GUI modules are in the example configuration. You can load a different configuration in the management window.

Check what hardware modules are around in the `hardware` directory.

See what other people are working on in the network graph https://github.com/Ulm-IQO/qudi/network .

In each Qudi module file there is a short example configuration snippet for that module.

Start by replacing the dummy hardware modules in the example configuration by real ones for hardware you have.

Neverhorst, 2019-09-06:
No actions needed

closed 2019-09-06 by Neverhorst