Ulm-IQO/qudi-legacy #548: prints and errors get lost when multiple qudi notebooks are open

kay-jahnke, 2019-09-17:
## *What* is affected by this bug?

This bug was introduced by making the qudi kernel deliver prints and errors in an asynchronous way. (see PR #539 #528)

## *When* does this occur?

start one jupyter notebook, and print something.
start a second notebook and also print something else.
redo the print in the first notebook -> it should appear in the second notebook.
If the second notebook hangs, any prints or errors are lost.

## *Where* on the platform does it happen?

any platform

## Expected behavior (i.e. solution)

The qudi kernel should figure out from which thread an output comes from and send that to the correct stream (notebook, stdout, stderr or maybe even a logger). As python has a single interpreter lock it also only has once hook for streams from different threads, making this problem non-trivial.
A start of a solution can be found in the branch https://github.com/Ulm-IQO/qudi/tree/fix_jupyter_multiple_kernels, but the really solve it the core needs to be rewritten to tag all streams in general and handle the output. Only kernel level does not suffice.

## Other Comments
This has been discussed in PR #539

alrik-durand, 2019-09-17:
Thanks for opening this issue.

I will test the PR to see how it behave on our setup. 

But if you say that restarting a notebook might cause trouble, then I'm worried about something :
I sometime have issues with plots, like if I plot in notebook 1, and then I want to plot something else in notebook 2, it usually does not show up. I've noticed that a simple trick is to click on restart on notebook 2, which does not affect the global context but makes it possible to plot on notebook 2 (disabling notebook 1 in the process). This is quite a dirty trick, maybe there is a better solution that I'm unaware of or maybe no one had had this problem before. If I can't restart the notebook safely, then this will become an issue to me !
Is is this a normal "bug" that would be hard to remove ?

kay-jahnke, 2019-09-19:
You are correct, in the current master version, the prints are always pulled to the most current notebook, so restarting the notebook will pull the focus to this notebook.
The branch solves this problem in a different ways, but the effect stated above should still be valid. 
If you lose outputs, please check the stdout or stderr of the qudi core (either your IDE or the console). They should appear there.