Ulm-IQO/qudi-legacy #552: Multi interface / Fastcounter problem

alrik-durand, 2019-10-17:
## *What* is affected by this bug?
We are trying to use the new multi interface feature  (#542) to use a hardware module as slow and fast counter. We have a working version, but for some mysterious reason, the activation fails about one every two time.

## *When* does this occur?
When we try to load the Pulsemeasurement GUI, and it tries to load all the sub modules. Sometime we activate this one first, sometime we activate the slow counter GUI before, there is no obvious correlation.

## *Where* on the platform does it happen?
We have observed this on Windows 10 with last version of Qudi

## *How* do we replicate the issue?
I created a branch called "timeharp260_dummy" with the basic code necessary to replicate the issue. I removed all the dll call so that it can be replicated on a computer without the physical hardware.
I also modified the default.cfg so that it is easy to test. Just load Qudi on this branch, then start pulsemeasurement GUI, and one every few time you get the error given bellow.

## Other Comments

The error given by Qudi is : 

`
Error during activation
KeyError: 'No method registered for interface "FastCounterInterface".'
Traceback (most recent call last):

  File "C:\Users\alrik\Documents\GitHub\qudi\core\module.py", line 96, in wrap_event
    base_event(*args, **kwargs)

  File "C:\Users\alrik\AppData\Local\Continuum\anaconda3\envs\qudi\lib\site-packages\fysom\__init__.py", line 311, in fn
    self.transition()

  File "C:\Users\alrik\AppData\Local\Continuum\anaconda3\envs\qudi\lib\site-packages\fysom\__init__.py", line 306, in _tran
    self._after_event(e)

  File "C:\Users\alrik\AppData\Local\Continuum\anaconda3\envs\qudi\lib\site-packages\fysom\__init__.py", line 338, in _after_event
    return getattr(self, fnname)(e)

  File "C:\Users\alrik\AppData\Local\Continuum\anaconda3\envs\qudi\lib\site-packages\fysom\__init__.py", line 95, in _callback
    return func(obj, *args, **kwargs)

  File "C:\Users\alrik\Documents\GitHub\qudi\core\module.py", line 216, in __load_status_vars_activate
    self.on_activate()

  File "C:\Users\alrik\Documents\GitHub\qudi\gui\pulsed\pulsed_maingui.py", line 191, in on_activate
    self._activate_analysis_ui()

  File "C:\Users\alrik\Documents\GitHub\qudi\gui\pulsed\pulsed_maingui.py", line 2256, in _activate_analysis_ui
    self._pa_apply_hardware_constraints()

  File "C:\Users\alrik\Documents\GitHub\qudi\gui\pulsed\pulsed_maingui.py", line 2289, in _pa_apply_hardware_constraints
    fc_constraints = self.pulsedmasterlogic().fast_counter_constraints

  File "C:\Users\alrik\Documents\GitHub\qudi\core\connector.py", line 62, in __getattribute__
    attr = getattr(self.obj, args[1])

  File "C:\Users\alrik\Documents\GitHub\qudi\logic\pulsed\pulsed_master_logic.py", line 311, in fast_counter_constraints
    return self.pulsedmeasurementlogic().fast_counter_constraints

  File "C:\Users\alrik\Documents\GitHub\qudi\core\connector.py", line 62, in __getattribute__
    attr = getattr(self.obj, args[1])

  File "C:\Users\alrik\Documents\GitHub\qudi\logic\pulsed\pulsed_measurement_logic.py", line 253, in fast_counter_constraints
    return self.fastcounter().get_constraints()

  File "C:\Users\alrik\Documents\GitHub\qudi\core\connector.py", line 64, in __getattribute__
    return attr[self.interface]

  File "C:\Users\alrik\Documents\GitHub\qudi\core\interface.py", line 77, in __getitem__
    raise KeyError('No method registered for interface "{0}".'.format(key))

`
I tried looking into it without success, it seem like the two run at activation may involved, or a nasty "self" error ? I a bit lost here :(

alrik-durand, 2020-02-18:
Hi,

I'm sorry to ask again but nobody answered me. Can I do something to help ?

Alrik

Neverhorst, 2020-06-02:
Fixed by PR #601

closed 2020-06-02 by Neverhorst