Ulm-IQO/qudi-legacy #580: after I initiate odmr scan the GUI freezes

zchen665, 2020-02-28:
## *What* is affected by this bug?

## *When* does this occur?

## *Where* on the platform does it happen?

## *How* do we replicate the issue?

## Expected behavior (i.e. solution)

## Other Comments

Neverhorst, 2020-04-07:
Hello @zchen665,
first of all I would really appreciate if people would try to stick closely to the issue template and do their best to give a detailed(!) bug report.

Regarding your issue:
ODMR GUI and logic modules are some of the oldest modules in qudi. As such they have been tested countless times in the lab. If they suddenly simply refuse to work, it can only be your custom "Berkeley nucleonics 835 wave generator" hardware module that's causing it.
Since this hardware module has not been contributed to the qudi project repository, we can not offer any support for that.

Just a wild guess into the blue: Some code in the hardware module is an infinite loop and/or some code is blocking the Qt main event loop.

liuyanulm, 2020-10-08:
> ## _What_ is affected by this bug?
> modules involved: a mw module for Berkeley nucleonics 835 wave generator
> odmr_logic
> 
> ## _When_ does this occur?
> when start scan in odmr module
> 
> ## _Where_ on the platform does it happen?
> no exception or error.
> sweep never starts and GUI freezes

Hi @zchen665 ,

As @Neverhorst  mentioned, this module has been tested with multiple microwave devices in countless experiments. It should be quite robust and universal. 
However, it does not mean all the microwave devices could be operated with this module. for example, some anritisu microwave devices do not support for the  'sweep mode'.
So, here goes my suggestions:
1. first of all, make sure that your hardware code is compatible with the qudi coding style/format, and the functions in your microwave hardware file work. (For this, you need to be sure that you comprehend the user manual of the microwave device thoroughly)
2. test the different modes, cw mode, list mode, and sweep mode.
3. since your microwave frequency never starts to sweep, check your trigger if it's functional or even it has a trigger (some microwave devices do not have trigger in).

Neverhorst, 2020-11-19:
Closed due to lack of a reaction/reply, a detailed bug report and any means to reproduce the bug.

closed 2020-11-19 by Neverhorst