Ulm-IQO/qudi-legacy #591: Crashes when trying to load drivers. Ubuntu 19.10, discreet graphics.

bynkchan, 2020-04-25:
## *What* is affected by this bug?

Program crashes when trying to load counters.  I am using Ubuntu 19.10 and Python 3.7.0.  My computer uses a Ryzen CPU alongside a Radeon graphics card (CPU doesn't have integrated graphics).

## *When* does this occur?

Error occurs when trying to load a counter.  It appears it can't load certain drivers.

## *Where* on the platform does it happen?

The program launches fine, but with these errors:
libGL error: MESA-LOADER: failed to open radeonsi (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri)
libGL error: failed to load driver: radeonsi
libGL error: MESA-LOADER: failed to open radeonsi (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri)
libGL error: failed to load driver: radeonsi
libGL error: MESA-LOADER: failed to open swrast (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri)
libGL error: failed to load driver: swrast
Could not get git repo because: /home/bynk/qudi/qudi-master/qudi-master
Task scannerLocationRefocus added!
Fatal Python error: Segmentation fault

## *How* do we replicate the issue?

It appears that this issue occurs when trying to utilize a discreet graphics card, since the program runs fine on my laptop with Intel integrated graphics.  To replicate the issue, I suppose you can run the program on any computer using discreet graphics, preferably with Ubuntu as the OS.

## Expected behavior (i.e. solution)

The counter should have loaded.

## Other Comments
I honestly do not know why the drivers won't load.  I have traced down the directories and found that the drivers exist.  No other programs have had any problems.  I've only been on Ubuntu for a few months, so I'm quite new.  Attached is a screenshot of the program upon launch as well as the error messages and the folder containing the drivers.  
[image]

Next is a screenshot of the error when the program crashes (because I tried loading a counter).
[image]

kay-jahnke, 2020-04-27:
@bynkchan Thank you for this issue.
It shows that other people are working with linux, which I much apprechiate.

I can confirm a similar behaviour in the following environment:
- Intel(R) Core(TM) i5-6200U CPU @ 2.30GHz
- GLX version: 1.4
- Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Intel (0x8086)
    Device: Mesa Intel(R) HD Graphics 520 (SKL GT2) (0x1916)
    Version: 20.0.5
    Accelerated: yes
    Video memory: 3072MB
    Unified memory: yes
    Preferred profile: core (0x1)
    Max core profile version: 4.6
    Max compat profile version: 4.6
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
- Linux 5.6.6-1-MANJARO
- Python 3.6.5 |Anaconda, Inc.| (default, Apr 29 2018, 16:14:56)
- pyqtgraph-qo = 0.10.0.4
- QtPy = 1.4.1
- `qmake --version`: QMake version 3.1; Using Qt version 5.14.2 in /usr/lib

Here I get the following error:
<code>
libGL error: MESA-LOADER: failed to open iris (search paths /usr/lib/dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open iris (search paths /usr/lib/dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open swrast (search paths /usr/lib/dri)
libGL error: failed to load driver: swrast
</code>

I my case however this does not prevent qudi from loading.

The error only appeared recently on my end. So I was rather suspecting the linux kernel or the qt-kernel to be the problem. As Ubuntu 19.10 is using a 5.3 it could still be from the kernel. I will try different kernels, to narrow down the problem.

kay-jahnke, 2020-04-27:
I checked the following kernel versions and they all show the same behavior to me:
- 5.6.7-1
- 5.4.35-1
- 4.19.118-1
- 4.14.177-1

From this I would conclude that it is not the kernel itself but probably something to do with the graphics driver or the way the pyqt accesses it through OpenGL.

Neverhorst, 2020-09-28:
Well, any news on this topic?
A quick google search reveals that this is a very general issue with a huge variety of applications on linux.
This is why I'm pretty sure it is a general problem with the display driver or related and not a problem tied to qudi.
Can we agree to close this PR until something is indicating a problem we can fix in qudi?

closed 2021-03-24 by Neverhorst