Ulm-IQO/qudi-legacy #593: Scientific Spinbox and Mapper are incompatible

tobiasgehring, 2020-04-29:
The mapper expects the widget to have a Qt Property for the value. However, scientific spinbox does not define `value` as a property as regular spinboxes do.

I'm not sure if it is possible to define `value` as Qt property in python as the getter has the same name. It's apparently a standard in Qt that they share the same name, but a quick look into the documentation did not reveal if this is possible within python. Probably not.

2 possible solutions:

1) add an attribute to the scientific spinbox, e.g. called `value_int` (or something else) which is the Qt property. This is a one line change per spinbox. This property could then be used with mapper. Naming is non-standard though as the property of a regular spinbox is called `value`.

2) Change mapper to also accept getter and setter instead of a Qt property and use this if mapper is used with the scientific spinbox.

My quick fix is 1, but I don't want to make a pull request without comments.

Neverhorst, 2020-06-23:
Well, my preferred solution would be to revise the scientific spinboxes to make use of `QtCore.Property` for all properties. This should not only solve the mapper issue but also get rid of some minor bugs when using these widgets in QtDesigner.

After a quick read on `QtCore.Property` in PyQt5 it seems that you can not use them as classic Python property which causes trouble for users/programmers not familiar with Qt.
However since we are moving to PySide2 this approach is becoming more appealing. There they state that a `QtCore.Property` functions as Qt AND Python property. (https://wiki.qt.io/Qt_for_Python_UsingQtProperties)

So depending on how fast you need it, I would suggest revising the scientific spinbox in the upcoming PySide2 qudi core and optionally fix it for now in the least invasive way possible (probably your suggestion no. 1)

See also: PR #605

Neverhorst, 2020-09-28:
Fixed with PR #605

closed 2020-09-28 by Neverhorst