Ulm-IQO/qudi-legacy #595: error on deleted StatusVariables

kay-jahnke, 2020-05-07:
## *What* is affected by this bug?

If a status variable is deleted at runtime and the responding module is deloaded by the manager, qudi tries to save status variables. For some reason deleted the status variable then repalced again by an object of type: core.statusvariable.StatusVar.
This causes an error when the value of this status variable is being tried to represent.

## *When* does this occur?

Delete a status variable of an arbitrary module (e.g. "test_variable") by `del module.test_variable`.

**Error:**
```
Failed to save status variables of module logic.df_logic:
OrderedDict([('save_raw_data', False), ('save_time_trace', False), ('sample_name', 'zone refined sample'), ('time_resolution', 0.0001), ('_adc_resolution', '16BIT'), ('signal_channel', 'A'), ('signal_voltage_range', '5V'), ('signal_offset', 4.5), ('trigger', 'EXT'), ('trigger_threshold', 500), ('hardware_oversampling', 10), ('lead_buffer', 0.02), ('trail_buffer', 0.0), ('averages', 10), ('led_percentage', 10), ('fit_offset', 0.001), ('testing_dicts', <core.statusvariable.StatusVar object at 0x7fb7e03f3f28>)])
Traceback (most recent call last):
  File "/home/kay/PycharmProjects/qudi/core/manager.py", line 1297, in saveStatusVariables
    config.save(filename, variables)
  File "/home/kay/PycharmProjects/qudi/core/config.py", line 250, in save
    ordered_dump(data, stream=f, Dumper=yaml.SafeDumper, default_flow_style=False)
  File "/home/kay/PycharmProjects/qudi/core/config.py", line 227, in ordered_dump
    return yaml.dump(data, stream, OrderedDumper, **kwds)
  File "/home/kay/miniconda/envs/qudi/lib/python3.6/site-packages/ruamel/yaml/main.py", line 849, in dump
    version=version, tags=tags, block_seq_indent=block_seq_indent)
  File "/home/kay/miniconda/envs/qudi/lib/python3.6/site-packages/ruamel/yaml/main.py", line 810, in dump_all
    dumper._representer.represent(data)
  File "/home/kay/miniconda/envs/qudi/lib/python3.6/site-packages/ruamel/yaml/representer.py", line 73, in represent
    node = self.represent_data(data)
  File "/home/kay/miniconda/envs/qudi/lib/python3.6/site-packages/ruamel/yaml/representer.py", line 101, in represent_data
    node = self.yaml_representers[data_types[0]](self, data)
  File "/home/kay/PycharmProjects/qudi/core/config.py", line 165, in represent_ordereddict
    dict_data.items())
  File "/home/kay/miniconda/envs/qudi/lib/python3.6/site-packages/ruamel/yaml/representer.py", line 206, in represent_mapping
    node_value = self.represent_data(item_value)
  File "/home/kay/miniconda/envs/qudi/lib/python3.6/site-packages/ruamel/yaml/representer.py", line 111, in represent_data
    node = self.yaml_representers[None](self, data)
  File "/home/kay/miniconda/envs/qudi/lib/python3.6/site-packages/ruamel/yaml/representer.py", line 375, in represent_undefined
    raise RepresenterError("cannot represent an object: %s" % data)
ruamel.yaml.representer.RepresenterError: cannot represent an object: <core.statusvariable.StatusVar object at 0x7fb7e03f3f28>
```

## *Where* on the platform does it happen?

Occurred on Linux 5.6.8-1-MANJARO with dummy, but should happen everywhere.

## *How* do we replicate the issue?

Delete a status variable of an arbitrary module (e.g. "test_variable") by `del module.test_variable`.

## Expected behavior (i.e. solution)

It should just not save the deleted status variable and on the next load of the module, the default value of this status variable should be restored.

## Other Comments

Neverhorst, 2020-05-07:
This is not an issue on the new core. 
Found out why it happens in the current master and it's easy to fix.
Btw... deleting class meta attributes at runtime is a big no-no.

kay-jahnke, 2020-05-08:
Ah, I understand that the moment you call del the variable del reverts back to its bare metaclass, interesting I had never though about that behavior.
Actually, as you remarked this can cause quite a bit of havoc not only during save and reload of the status variable (e.g. if status_variable would always return true on a deleted status_variable). 

Is there an easy way to avoid the del and rather through an error at that point? Not that it needs to be fix, just for my curiosity.

Neverhorst, 2020-05-08:
> Ah, I understand that the moment you call del the variable del reverts back to its bare metaclass, interesting I had never though about that behavior.

Upon activation the module __instance__ gets the value of the `StatusVar` object (either default or loaded from app_status) attached as attribute. 
Since the attribute name is the same as the __class__ attribute holding the `StatusVar` object instance, `self.var_name` will return the __instance__ attribute (value) and not the __class__ attribute (`StatusVar` instance). 
Of course if you delete now the instance attribute (by e.g. `del self.var_name` or `del my_module_name.var_name`) the lookup of the attribute name will fall back to the class attribute instead since the instance does not have it anymore. 

> Actually, as you remarked this can cause quite a bit of havoc not only during save and reload of the status variable (e.g. if status_variable would always return true on a deleted status_variable). [...]

Yes, you could implement `__delattr__(self, name)` in `ModuleBase` and check for "forbidden" attribute names to delete. If you want to prevent deletion, raise an `AttributeError`.

closed 2020-05-08 by Neverhorst