Ulm-IQO/qudi-legacy #611: error: Timeout when loading remote hardware module

meiwud334, 2020-08-26:
## *What* is affected by this bug?

Timeout error occurs when attempting to remotely access the hardware module nicard (by defining it as a remote module in config via rpyc).

## *When* does this occur?

Timeout error occurs when attempting to remotely access the hardware module nicard (by defining it as a remote module in config via rpyc).

## *Where* on the platform does it happen?

Error while loading hardware module: remotenicard
socket.timeout: timed out
Traceback (most recent call last):

  File "D:\Qudi_git\qudi\core\manager.py", line 771, in loadConfigureModule
    keyfile=keyfile)

  File "D:\Qudi_git\qudi\core\remote.py", line 160, in getRemoteModuleUrl
    return self.getRemoteModule(parsed.hostname, parsed.port, name)

  File "D:\Qudi_git\qudi\core\remote.py", line 173, in getRemoteModule
    module = RemoteModule(host, port, name, certfile=certfile, keyfile=keyfile)

  File "D:\Qudi_git\qudi\core\remote.py", line 228, in __init__
    self.connection = rpyc.connect(host, port, config={'allow_all_attrs': True})

  File "C:\Users\ou\Anaconda3\envs\qudi\lib\site-packages\rpyc\utils\factory.py", line 93, in connect
    s = SocketStream.connect(host, port, ipv6=ipv6, keepalive=keepalive)

  File "C:\Users\ou\Anaconda3\envs\qudi\lib\site-packages\rpyc\core\stream.py", line 154, in connect
    return cls(cls._connect(host, port, **kwargs))

  File "C:\Users\ou\Anaconda3\envs\qudi\lib\site-packages\rpyc\core\stream.py", line 112, in _connect
    s.connect(sockaddr)

## *How* do we replicate the issue?

set remote access true on host pc and added it as a remote hardware module on client pc.

## Expected behavior (i.e. solution)

When try to load the shared hardware on client pc 

## Other Comments

Neverhorst, 2020-08-27:
Well, I see no reason why this should be an issue specifically with NIcard.
Since you did not really give a lot of information regarding this issue, I would assume you simply have an erroneous setup of the remote connection. Probably you would get the same error when trying to connect to a different module.

meiwud334, 2020-08-27:
Yes the descriptions were a bit off, it was not trying to specify the problem for ni card... Anyway it turned out to be a new version of pythom vs. old version of config file proble and was solved. Thanks for the reply:).

closed 2020-08-27 by meiwud334