Ulm-IQO/qudi-legacy #618: Provided GitHub link in qudi manager runs in page not found error

ksenkalla, 2020-10-07:
## *What* is affected by this bug?
The link shown in the bottom of the qudi manager, which should show the used version of qudi, runs to an 404 error, page not found error on GitHub website.

## *When* does this occur?
It occurs with the newest qudi version. (Maybe also for some newer versions than >Apr 2020)

## *Where* on the platform does it happen?

## *How* do we replicate the issue?
Just click on the provided link in the bottom of the manager gui.

## Expected behavior (i.e. solution)
Usually it shows the latest changes on GitHub website.

## Other Comments

liuyanulm, 2020-10-07:
Hi @ksenkalla , 
My qudi shows the lates update of #617 
I could not reproduce this issue.
one small possiblity, maybe your computer was offline?

ksenkalla, 2020-10-07:
Sorry, this was an user problem. It occured because I had commits on local master branch which automatically should be synchronized on all roots.
To solve this I had to change settings on pycharm "execute branch operations on all roots" and reverse unwanted commits.

closed 2020-10-07 by ksenkalla