Ulm-IQO/qudi-legacy #663: Qudi Confocal Scanner with NI USB-6212 DAQ Card

eleanornichols, 2021-06-15:
## *What* is affected by this bug?

I'm trying to use the NI USB-6212 DAQ card with Qudi to be able to do confocal scans. This DAQ card has just 2 analogue outputs which we want to use with a scanning mirror. We are not interested in doing z scans.  I am also using the national instruments x series code for the DAQ card (although I am aware that I am not using an x series)

Is it possible to do just xy scans using the confocal module, and not have to connect an analogue output for z? Has anyone faced this issue before?

## *When* does this occur?

When I load Qudi, I get an error saying "Error during activation. IndexError: List index out of range" which I believe is referring to the confocal module needing 3 analogue outputs while I only have 2 (and only put 2 in the config file). I get the same error when I try and open the confocal module. 

## *Where* on the platform does it happen?

When I load Qudi or the confocal module

## *How* do we replicate the issue?

Have a DAQ card with 2 analogue outputs. Assign these 2 analogue outputs to scanner_ao_channels in the config file. Try and run Qudi.

## Expected behavior (i.e. solution)

I was hoping that I would be able to run xy scans, but I cannot. 

## Other Comments

LukePiewalker, 2021-06-18:
Dear @eleanornichols,

Thanks for raising this issue. This is actually a very well known one. The whole confocal toolchain was one of the first qudi modules written and therefore is rather immature with regard to the hardware modularity which qudi tries to provide (see the [qudi paper](https://doi.org/10.1016/j.softx.2017.02.001)). The whole logic is pretty much tied to a 4 AO channel Ni Card scheme, which shouldn't be.

Unfortunately, I have no quick fix for you. You could try writing a logic an gui module which suits your needs but thats a rather tedious task. I think it might be possible to use the hardware file for the X Series with your Ni Daq, at least I could start it with an NI Max simulated device of your types after some tweaks of the config. Correct me if I'm wrong @QTslav @Neverhorst?

The good news are:
As we are aware of this issue, there are fixes on their way already. Qudi is currently undergoing some rather big rework such that it is based on "Qt for Python" (`PySide`) instead of `PyQt5` and the whole scanning-probing type of measurements will be (hopefully) as modular as qudi is intended to be once this rework is rolled out.

liuyanulm, 2022-03-18:
> ## _What_ is affected by this bug?
> I'm trying to use the NI USB-6212 DAQ card with Qudi to be able to do confocal scans. This DAQ card has just 2 analogue outputs which we want to use with a [...]

Hi @eleanornichols ,

The error message your povided is not pointing to any specific issue of the X serie code. I could assume this might be caused by other issues, not only to your 2AOs, for example, USB-6212 has two counters while typically qudi supports DAQ with 4 counters. 
To solve your issue, either you buy a recommended DAQ of X series, or you wrote your own hardware code for the USB6212. The first one saves you tons of time, with the double price of the USB-6212.

Good luck with your work!
Yan Liu