Ulm-IQO/qudi-legacy #698: Visa resource manager communication errors

pikestefan, 2022-02-08:
## *What* is affected by this bug?
Hello, I have forked qudi in July 2020. I have found an issue when loading the microwave.mw_source_srssg.MicrowaveSRSSG class.

## *When* does this occur?
When I load the hardware module, I get the error pasted at the end of this message. With a bit of debugging I found out that the issue occurs whenever the visa.ResourceManager() does any type of query (e.g. listing resources or connecting to a device).

## *How* do we replicate the issue?
I managed to reproduce the error with this fake hardware module:
```
from core.module import Base
import visa

class testerror(Base):
    def on_activate(self):
        self.res_manager = visa.ResourceManager()
        self.res_manager.list_resources()

    def on_deactivate(self):
       pass
```

## Error message
```
Windows fatal exception: access violation

Thread 0x0000330c (most recent call first):
  File "C:\Users\QMPL\Documents\GitHub\qudi\core\parentpoller.py", line 132 in run
  File "C:\Users\QMPL\Anaconda3\envs\qudi\lib\threading.py", line 916 in _bootstrap_inner
  File "C:\Users\QMPL\Anaconda3\envs\qudi\lib\threading.py", line 884 in _bootstrap

Thread 0x00002ba4 (most recent call first):
  File "C:\Users\QMPL\Anaconda3\envs\qudi\lib\site-packages\zmq\utils\garbage.py", line 47 in run
  File "C:\Users\QMPL\Anaconda3\envs\qudi\lib\threading.py", line 916 in _bootstrap_inner
  File "C:\Users\QMPL\Anaconda3\envs\qudi\lib\threading.py", line 884 in _bootstrap

Thread 0x00002a54 (most recent call first):
  File "C:\Users\QMPL\Anaconda3\envs\qudi\lib\site-packages\tornado\platform\select.py", line 63 in poll
  File "C:\Users\QMPL\Anaconda3\envs\qudi\lib\site-packages\tornado\ioloop.py", line 863 in start
  File "C:\Users\QMPL\Anaconda3\envs\qudi\lib\site-packages\ipykernel\iostream.py", line 78 in _thread_main
  File "C:\Users\QMPL\Anaconda3\envs\qudi\lib\threading.py", line 864 in run
  File "C:\Users\QMPL\Anaconda3\envs\qudi\lib\threading.py", line 916 in _bootstrap_inner
  File "C:\Users\QMPL\Anaconda3\envs\qudi\lib\threading.py", line 884 in _bootstrap

Thread 0x00000f64 (most recent call first):
  File "C:\Users\QMPL\Anaconda3\envs\qudi\lib\threading.py", line 295 in wait
  File "C:\Users\QMPL\Anaconda3\envs\qudi\lib\threading.py", line 551 in wait
  File "C:\Users\QMPL\Anaconda3\envs\qudi\lib\site-packages\IPython\core\history.py", line 829 in run
  File "C:\Users\QMPL\Anaconda3\envs\qudi\lib\site-packages\IPython\core\history.py", line 58 in needs_sqlite
  File "<decorator-gen-24>", line 2 in run
  File "C:\Users\QMPL\Anaconda3\envs\qudi\lib\threading.py", line 916 in _bootstrap_inner
  File "C:\Users\QMPL\Anaconda3\envs\qudi\lib\threading.py", line 884 in _bootstrap

Thread 0x00000c48 (most recent call first):
  File "C:\Users\QMPL\Anaconda3\envs\qudi\lib\socket.py", line 205 in accept
  File "C:\Users\QMPL\Anaconda3\envs\qudi\lib\site-packages\rpyc\utils\server.py", line 139 in accept
  File "C:\Users\QMPL\Anaconda3\envs\qudi\lib\site-packages\rpyc\utils\server.py", line 263 in start
  File "C:\Users\QMPL\Documents\GitHub\qudi\core\remote.py", line 252 in run

Current thread 0x00000358 (most recent call first):
  File "C:\Users\QMPL\Anaconda3\envs\qudi\lib\site-packages\pyvisa\ctwrapper\functions.py", line 485 in find_resources
  File "C:\Users\QMPL\Anaconda3\envs\qudi\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 205 in list_resources
  File "C:\Users\QMPL\Anaconda3\envs\qudi\lib\site-packages\pyvisa\highlevel.py", line 1571 in list_resources
  File "C:\Users\QMPL\Documents\GitHub\qudi\hardware\microwave\test_errors.py", line 14 in on_activate
  File "C:\Users\QMPL\Documents\GitHub\qudi\core\module.py", line 223 in __load_status_vars_activate
  File "C:\Users\QMPL\Anaconda3\envs\qudi\lib\site-packages\fysom\__init__.py", line 95 in _callback
  File "C:\Users\QMPL\Anaconda3\envs\qudi\lib\site-packages\fysom\__init__.py", line 338 in _after_event
  File "C:\Users\QMPL\Anaconda3\envs\qudi\lib\site-packages\fysom\__init__.py", line 306 in _tran
  File "C:\Users\QMPL\Anaconda3\envs\qudi\lib\site-packages\fysom\__init__.py", line 311 in fn
  File "C:\Users\QMPL\Documents\GitHub\qudi\core\module.py", line 98 in wrap_event
  File "C:\Users\QMPL\Documents\GitHub\qudi\core\manager.py", line 969 in activateModule
  File "C:\Users\QMPL\Documents\GitHub\qudi\core\manager.py", line 1172 in startModule
  File "C:\Users\QMPL\Documents\GitHub\qudi\gui\manager\managergui.py", line 622 in loadButtonClicked
  File "C:\Users\QMPL\Documents\GitHub\qudi\core\__main__.py", line 311 in <module>
  File "C:\Users\QMPL\Anaconda3\envs\qudi\lib\runpy.py", line 85 in _run_code
  File "C:\Users\QMPL\Anaconda3\envs\qudi\lib\runpy.py", line 193 in _run_module_as_main

```

Neverhorst, 2022-02-09:
Hi @pikestefan. 
Your example illustrates that your issue has nothing to do with _qudi_ itself. Apparently your `pyvisa` package installation or the used system visa binary installation is somehow broken.
I bet you can reproduce the same problem just running the code in the same Python environment (without _qudi_).

pikestefan, 2022-02-09:
Hi @Neverhorst,
Do you mean I should get the same error when trying to use the resource manager within the qudi environment? I gave it a quick try in the anaconda prompt, but nothing looks amiss there.
[image]

pikestefan, 2022-03-01:
Thanks a lot for your reply! It turns out the problem was indeed the VISA library itself: after uninstalling and reinstalling the distribution, the error stopped popping up.

closed 2022-03-01 by pikestefan

Neverhorst, 2022-03-01:
Glad everything worked out.