Ulm-IQO/qudi-legacy #710: qudiinstallation, qudi-core vs qudi

AItah, 2023-10-30:
Dear all,

When following qudi installation (https://github.com/Ulm-IQO/qudi/blob/master/documentation/installation.md) it ends up with Python 3.6.5 Anaconda, Inc. However, in qudi-core it said to support 3.8 to 3.10 python version.

1. Can we upgrade the python version from 3.6.5?
2. What is the different between qudi-core and qudi?

Kind Regards,
Amir

timoML, 2023-10-30:
Please note the very first line of the readme file in this repo:

⚠ DEPRECATION WARNING

 The development of this qudi repository has ceased. We strongly recommend migrating to our new [qudi-core project](https://github.com/Ulm-IQO/qudi-core) that is actively developed. Framework functionality is now separated from the measurement toolchains. Most features of the old project can be recovered by installing the qudi-core and qudi-iqo-modules packages.

If you're running the old qudi (< v0.1), you cannot upgrade Python beyond 3.6.

Neverhorst, 2023-10-30:
Hi @AItah ,

This `qudi` repository is deprecated and is not under active development anymore.
All our efforts are focused around `qudi-core` (the base application and framework) and in extension `qudi-iqo-modules` (measurement toolchains of the Ulm University Institute for Quantum Optics and external contributors).

Please ask all general `qudi` questions [here](https://github.com/Ulm-IQO/qudi-core/discussions/)
and all questions related to measurement modules from `qudi-iqo-modules` [here](https://github.com/Ulm-IQO/qudi-iqo-modules/discussions)
Information about the "new qudi" can be found on the [`qudi-core` repository homepage](https://github.com/Ulm-IQO/qudi-core)

closed 2023-10-30 by Neverhorst

AItah, 2023-10-30:
thank you for the information. I will use qudi-core.